PRINTR3D/formide-client #90: Update slice profile for integration with Katana v0.9.0-alpha

karimcitoh, 2015-08-17:
In the following document are listed the latest changes that should be made in the slice profiles:

https://docs.google.com/a/printr.nl/spreadsheets/d/1CcFjNGSTZEkmu3FSvioaTRSom7Gope-ScKhb4phixlc/edit?usp=sharing

This document only shows changes in format. If you want to know more about the new functionality, more documentation will be added soon to the wiki!

ChrisTerBeke, 2015-09-03:
Will do asap. same structure as done in the cloud using the reference file

karimcitoh, 2015-09-03:
Can we not use the same file?

ChrisTerBeke, 2015-09-03:
that's what I said ;) I'll get the latest reference file that fits the shipped katana binary for the client and add that to the repo

ChrisTerBeke, 2015-09-03:
Or maybe you can update the file in the repo when you push a new katana binary to it

karimcitoh, 2015-09-03:
Updating the file in the repo sounds like the best way of updating this file. So we can also visualize changes with commits

ChrisTerBeke, 2015-09-04:
cool. I'll add the current file when I implement this and from there you can update it together with the binary.

closed 2015-09-10 by ChrisTerBeke