PRINTR3D/formide-client #98: Using wrong npm error variable

cb1kenobi, 2015-09-25:
Noticed this while looking for programmatic examples of using npm and noticed when an npm call fails, you're passing the wrong error variable. There's a handful of places that have the same issue.

```
npm.commands.install([packageName], function (installErr, data) {
    if (installErr) return cb(err); // <----------  should be installErr
```

https://github.com/PRINTR3D/formide-client/blob/master/core/modules/update/index.js#L65
https://github.com/PRINTR3D/formide-client/blob/master/core/modules/update/index.js#L87
https://github.com/PRINTR3D/formide-client/blob/master/core/modules/update/index.js#L105
https://github.com/PRINTR3D/formide-client/blob/master/core/modules/update/index.js#L122
https://github.com/PRINTR3D/formide-client/blob/master/core/modules/update/index.js#L141

ChrisTerBeke, 2015-09-25:
Thanks! Will push a fix in the next release

ChrisTerBeke, 2015-09-30:
Fixed in npm pushed version 0.2.33

closed 2015-09-30 by ChrisTerBeke