PRINTR3D/formide-client #100: Cloud calls not working (homing, heating temperature, ...)

karimcitoh, 2015-09-28:
When I use commands using the local interface, it works.

But when I use the cloud interface, I receive the calls in The Element but the driver functions are not called. This is the output:
```
12:02:27:764	 [core/modules/cloud/index.js]............."Cloud http call: api/printer/ttyUSB1/home"
12:02:43:546	 [core/modules/cloud/index.js]............."Cloud http call: api/printer/ttyUSB1/home"
12:02:44:479	 [core/modules/cloud/index.js]............."Cloud http call: api/db/queue"
12:02:51:349	 [core/modules/cloud/index.js]............."Cloud http call: api/printer/ttyUSB1/home"
```

ChrisTerBeke, 2015-09-28:
wtf, never had this before. Are you sure the cloud module is correctly connected to the local http server and such (what do you see when you boot formide-client?)

karimcitoh, 2015-09-28:
```
13:26:52:338	 [bootstrap.js]............................"=========================================="
13:26:52:391	 [bootstrap.js]............................"        Starting FORMIDEOS v0.2.26"
13:26:52:399	 [bootstrap.js]............................"=========================================="
13:26:52:457	 [core/utils/moduleManager.js]............."module process loaded"
13:26:56:550	 [core/utils/moduleManager.js]............."module db loaded"
13:26:56:565	 [core/utils/moduleManager.js]............."module settings loaded"
13:26:56:594	 [core/utils/moduleManager.js]............."module auth loaded"
13:26:57:41	 [core/utils/moduleManager.js]............."module log loaded"
13:26:58:760	 [core/utils/moduleManager.js]............."module files loaded"
13:26:58:790	 [core/utils/moduleManager.js]............."module printer loaded"
13:26:58:866	 [core/utils/moduleManager.js]............."module slicer loaded"
13:26:58:991	 [core/utils/moduleManager.js]............."module setup loaded"
13:27:01:84	 [core/utils/moduleManager.js]............."module update loaded"
13:27:01:772	 [core/utils/moduleManager.js]............."module cloud loaded"
13:27:03:439	 [core/utils/moduleManager.js]............."module formide-client-interface loaded"
13:27:05:559	 [core/utils/moduleManager.js]............."module formide-client-webhook loaded"
13:27:05:579	 [core/utils/moduleManager.js]............."module process activated"
13:27:06:182	 [core/utils/moduleManager.js]............."module db activated"
13:27:06:203	 [core/utils/moduleManager.js]............."module settings activated"
13:27:06:250	 [core/utils/moduleManager.js]............."module auth activated"
13:27:06:330	 [core/utils/moduleManager.js]............."module log activated"
13:27:06:888	 [core/utils/moduleManager.js]............."module files activated"
13:27:06:946	 [core/modules/printer/index.js]..........."Binded Formidriver in rpi/Formidriver"
13:27:06:954	 [core/modules/printer/index.js]..........."Formidriver started successfully"
13:27:06:979	 [core/utils/moduleManager.js]............."module printer activated"
13:27:07:20	 [core/modules/slicer/index.js]............"Binded katana in rpi/katana"
13:27:07:38	 [core/utils/moduleManager.js]............."module slicer activated"
13:27:07:59	 [core/utils/moduleManager.js]............."module setup activated"
13:27:07:116	 [core/utils/moduleManager.js]............."module update activated"
13:27:07:562	 [core/utils/moduleManager.js]............."module cloud activated"
13:27:07:574	 [odules/formide-client-interface/index.js].."Interface started on port 8080"
13:27:07:607	 [core/utils/moduleManager.js]............."module formide-client-interface activated"
13:27:07:625	 [core/utils/moduleManager.js]............."module formide-client-webhook activated"
13:27:07:635	 [core/utils/http/index.js]................"server running on port 1337"
13:27:07:756	 [core/modules/process/index.js]..........."Error: listen EADDRNOTAVAIL\n    at exports._errnoException (util.js:746:11)\n    at Server._listen2 (net.js:1139:19)\n    at listen (net.js:1182:10)\n    at net.js:1280:9\n    at dns.js:85:18\n    at process._tickCallback (node.js:355:11)\n    at Function.Module.runMain (module.js:503:11)\n    at startup (node.js:129:16)\n    at node.js:814:3"
+++++ printer +++++ 
sending event: printer connected
NEW MARLIN DRIVER
SETTING STATE: 1
new driver for printer: /dev/ttyUSB0
SETTING STATE: 1
13:27:08:214	 [_modules/formide-client-webhook/index.js].."Could not send to webhook"
13:27:08:387	 [core/modules/printer/index.js]..........."Printer connected: /dev/ttyUSB0"
13:27:09:847	 [core/modules/cloud/index.js]............."Cloud connected"
SETTING STATE: 1
SETTING STATE: 1
13:27:10:282	 [core/modules/cloud/index.js]............."Cloud syncPrinters"
SETTING STATE: 1
13:27:10:676	 [core/modules/cloud/index.js]............."Synced printer from cloud: 1"
13:27:10:700	 [core/modules/cloud/index.js]............."Synced printer from cloud: 1"
checking connection
13:27:12:590	 [core/modules/cloud/index.js]............."Cloud user authorized with access_token 0c30090e-75a0-48ec-97bb-61e5f206af12"
checking connection
13:27:14:146	 [core/modules/cloud/index.js]............."Cloud http call: api/db/queue"
checking connection
SETTING STATE: 2
switching baudrate to 250000
checking connection
SETTING STATE: 3
SETTING STATE: 3
13:27:19:390	 [core/modules/printer/index.js]..........."Printer online: /dev/ttyUSB0"
13:28:28:116	 [core/modules/cloud/index.js]............."Cloud http call: api/db/queue"
13:28:52:863	 [core/modules/cloud/index.js]............."Cloud user authorized with access_token e876e781-3d75-4e7e-98a9-4652fa1ce1b9"
13:28:53:96	 [core/modules/cloud/index.js]............."Cloud http call: api/update/core"
13:29:01:258	 [core/modules/cloud/index.js]............."Cloud http call: api/printer/ttyUSB0/home"

```

This is what the output looks like. Could it be related to #99 ? Or maybe to the error EADDRNOTAVAIL shown above?

ChrisTerBeke, 2015-09-28:
```
13:27:07:756     [core/modules/process/index.js]..........."Error: listen EADDRNOTAVAIL\n    at exports._errnoException (util.js:746:11)\n    at Server._listen2 (net.js:1139:19)\n    at listen (net.js:1182:10)\n    at net.js:1280:9\n    at dns.js:85:18\n    at process._tickCallback (node.js:355:11)\n    at Function.Module.runMain (module.js:503:11)\n    at startup (node.js:129:16)\n    at node.js:814:3"
```

looks like that something else is using this port?

ChrisTerBeke, 2015-09-30:
All seems to work (tested on OSX and Element)

closed 2015-09-30 by ChrisTerBeke