PRINTR3D/formide-client #107: Client crashing after adding gcode to queue (cloud)

karimcitoh, 2015-10-15:
```
16:30:25:672	 [core/modules/cloud/index.js]............."Cloud addToQueue: HfzYEhQE2kgcyIfVwNKJq3l2SzOBgvlMNDNEDLJAXHpdrX9srZ"
/home/debian/npm-global/lib/node_modules/formide-client/core/modules/cloud/index.js:183
				printer: printer.toObject()
				                ^
TypeError: Cannot read property 'toObject' of null
    at Promise.<anonymous> (/home/debian/npm-global/lib/node_modules/formide-client/core/modules/cloud/index.js:183:21)
    at Promise.<anonymous> (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mpromise/lib/promise.js:177:8)
    at Promise.emit (events.js:107:17)
    at Promise.emit (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mpromise/lib/promise.js:84:38)
    at Promise.fulfill (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mpromise/lib/promise.js:97:20)
    at Promise.resolve (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/lib/promise.js:114:23)
    at Promise.<anonymous> (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mpromise/lib/promise.js:177:8)
    at Promise.emit (events.js:107:17)
    at Promise.emit (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mpromise/lib/promise.js:84:38)
    at Promise.fulfill (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mpromise/lib/promise.js:97:20)
    at Immediate.cb (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/lib/query.js:1146:30)
    at Immediate._onImmediate (/home/debian/npm-global/lib/node_modules/formide-client/node_modules/mongoose/node_modules/mquery/lib/utils.js:137:16)
    at processImmediate [as _immediateCallback] (timers.js:367:17)
```

ChrisTerBeke, 2015-10-15:
Ah, probably because we disabled cloud/local printer syncing (it caused some issues). Will try to fix tonight (maybe use just the port of a printer).

ChrisTerBeke, 2015-10-15:
Fixed by using port only to bind queue items to a printer

closed 2015-10-15 by ChrisTerBeke