PRINTR3D/formide-client #117: Server Error Printer Control

regnerus, 2015-12-08:
When homing, moving etc. or setting the temperature it _does_ execute the command on the printer but the server return a `500 server error`

```
{
  "statusCode": 500,
  "statusName": "Server Error",
  "message": "response is not defined",
  "error": {}
}
```

ChrisTerBeke, 2015-12-08:
any other stuff in the terminal?

regnerus, 2015-12-08:
```
debug: debug: Cloud authenticate user:56421e2160d748fe0239b639
debug: debug: Cloud user authorized with access_token bab2e323-01f3-4677-8c29-725e41e7e9a8
debug: debug: Cloud http call: api/printer/tty.usbserial-A70392XK/home
Sending raw command
Response from writing: ok 0
count:1
```

ChrisTerBeke, 2015-12-08:
reminder: this repo is open, maybe not post access tokens :P

regnerus, 2015-12-08:
Haha just thought of that too. :fearful:

commit 8d9fbc23 (2015-12-08): fix #117
closed 2015-12-08 by ChrisTerBeke