PRINTR3D/formide-client #129: HTTP Slice Request: Cannot read property 'undefined' of undefined

regnerus, 2015-12-14:
When sending a HTTP slice request the server responds with 500 server error, this has something to do with the override settings materials array. It is in the format that Dorien supplied me with but maybe this has not been updated in the client yet?

Example of my request:

```
{
	"files": [183],
	"sliceProfile": 1,
	"materials": [1],
	"printer": 1,
	"settings": {
		"brim": {
			"use": false,
			"extruder": 0
		},
		"raft": {
			"use": false,
			"extruder": 0
		},
		"support": {
			"use": false,
			"extruder": 0
		},
		"skirt": {
			"use": true,
			"extruder": 0
		},
		"fan": {
			"use": true,
			"speed": 100
		},
		"bed": {
			"use": true,
			"temperature": 60,
			"firstLayersTemperature": 65
		},
		"files": [{
			"extruder": 0,
			"position": {
				"x": 100000,
				"y": 100000,
				"z": 20000
			},
			"rotation": {
				"x": 0,
				"y": 0,
				"z": 0
			},
			"scale": {
				"x": 1,
				"y": 1,
				"z": 1
			}
		}],
		"override": {
			"fan": {
				"fullSpeedLayer": 2,
				"min": 35,
				"max": 100,
				"smart": false,
				"factor": 40,
				"minimalSeconds": 23
			},
			"raft": {
				"baseThickness": 0,
				"baseLineWidth": 1000,
				"baseSpeed": 40,
				"interfaceThickness": 250,
				"interfaceLineWidth": 400,
				"surfaceThickness": 250,
				"surfaceLineWidth": 400,
				"surfaceSpeed": 60,
				"surfaceLayers": 1,
				"airGap": 200,
				"firstLayerAirGap": 250,
				"extraMargin": 2000,
				"fanSpeed": 25
			},
			"infill": {
				"amount": 10,
				"overlap": 90,
				"pattern": "INFILL_GRID"
			},
			"support": {
				"amount": 10,
				"activationAngle": 60,
				"orientationAngle": 0,
				"wallDistance": 1000000,
				"supportEverywhere": false,
				"supportPlatform": false,
				"supportModel": false,
				"pattern": 0,
				"XYDistance": 900,
				"ZDistance": 150
			},
			"skirt": {
				"lineCount": 2,
				"distance": 6000,
				"minLength": 50000
			},
			"movement": {
				"combing": false,
				"oozeShield": false,
				"printSpeed": 80,
				"travelSpeed": 80,
				"innerWallSpeed": 70,
				"outerWallSpeed": 60,
				"infillSpeed": 80,
				"speedupLayers": 10,
				"smart": true,
				"minimalSeconds": 20,
				"minFactor": 40,
				"maximalSeconds": 60,
				"maxFactor": 0
			},
			"bottom": {
				"numberOfLayers": 2,
				"layerHeight": 200,
				"speed": 30,
				"solid": true,
				"layerDelay": 0,
				"firstLayersWidthFactor": 120
			},
			"top": {
				"numberOfLayers": 2,
				"layerHeight": 200,
				"solid": true
			},
			"retraction": {
				"amount": 5000,
				"speed": 30,
				"extruderSwitch": 14500,
				"minimalExtrusionBeforeRetraction": 400,
				"zHop": 0,
				"minimalDistance": 15000
			},
			"multiExtrusion": {
				"preSwitchCode": [],
				"postSwitchCode": [],
				"dualOverlap": 1500
			},
			"layers": {
				"spiralize": false,
				"simpleMode": false,
				"layerHeight": 300,
				"layerCoolingTime": 0,
				"firstLayersAmount": 100,
				"numberOutlines": 2,
				"solidLayers": 0,
				"sequentialMode": false,
				"wipeTowerSize": 3000
			},
			"gcode": {
				"changeLayerGcode": [],
				"customGcode": [],
				"repeatEvery": 0
			},
			"brim": {
				"lines": 4
			},
			"skin": {
				"skin": true,
				"infillOverlap": 20,
				"infillAmount": 100,
				"upSkinSpeed": 60,
				"downSkinSpeed": 60,
				"upSkinCount": 2,
				"downSkinCount": 2
			},
			"materials": [{
				"type": "PLA",
				"temperature": 195,
				"firstLayersTemperature": 205,
				"bedTemperature": 60,
				"feedRate": 100
			}]
		}
	}
}
```

ChrisTerBeke, 2015-12-14:
Probably a tools issue since I didn't have this before and didn't touch the client

closed 2015-12-14 by ChrisTerBeke

regnerus, 2015-12-14:
I'll ask Dorien then.