PRINTR3D/formide-client #146: Cura Import: Callback not found

regnerus, 2015-12-17:
`/db/sliceprofiles/cura` results in a `500 error`.

```
AssertionError: Callback is not found
   at Object.module.exports [as createSliceprofileFromCura] (/Users/Bouke/Sites/katana-desktop/node_modules/formide-tools/lib/createsliceprofilefromcura.js:33:3)
   at /Users/Bouke/Sites/katana-desktop/node_modules/formide-client/core/modules/db/rest/sliceprofile.js:90:16
   at Layer.handle [as handle_request] (/Users/Bouke/Sites/katana-desktop/node_modules/express/lib/router/layer.js:95:5)
   at next (/Users/Bouke/Sites/katana-desktop/node_modules/express/lib/router/route.js:131:13)
   at Form.<anonymous> (/Users/Bouke/Sites/katana-desktop/node_modules/connect-multiparty/index.js:101:9)
   at emitNone (events.js:72:20)
   at Form.emit (events.js:166:7)
   at maybeClose (/Users/Bouke/Sites/katana-desktop/node_modules/multiparty/index.js:557:10)
   at endFlush (/Users/Bouke/Sites/katana-desktop/node_modules/multiparty/index.js:552:3)
   at WriteStream.<anonymous> (/Users/Bouke/Sites/katana-desktop/node_modules/multiparty/index.js:617:5)
   at emitNone (events.js:72:20)
   at WriteStream.emit (events.js:166:7)
   at fs.js:1787:14
   at /Users/Bouke/Sites/katana-desktop/node_modules/graceful-fs/graceful-fs.js:42:10
   at FSReqWrap.oncomplete (fs.js:82:15)
```

ChrisTerBeke, 2015-12-17:
As you can see in the error, it's triggered in the formide-tools package, but I'll investigate.

ChrisTerBeke, 2015-12-17:
Fixed

closed 2015-12-17 by ChrisTerBeke