PRINTR3D/formide-client #281: Add auto-boot scripts for different Linux versions

ChrisTerBeke, 2017-02-23:
We should add some scripts that will make the client automatically boot when starting the device.

shaiss, 2017-02-23:
Here's what I have that sometimes works.   Note that export NODE_ENV=production works fine, it's the formide-client that fails to start.  I've tried putting this bash script in /etc/init.d/ and /etc/network/if-up.d/ to no avail

`export NODE_ENV=production
sudo /home/pi/.nvm/versions/node/v4.7.0/bin/formide-client
`

ChrisTerBeke, 2017-02-23:
Which version of Linux are your running? I'm personally on a version that uses `systemd` instead of `upstart`, but we wanna make this script work on both.

shaiss, 2017-02-23:
The latest Jessie lite download from raspberrypi.org.  My guess is it's
systemd

ChrisTerBeke, 2017-02-23:
With `systemd`, you can try the following:

Create a file `/etc/systemd/system/formide-client.service` and fill it with these contents:

```
[Unit]
Description=Formide Client

[Service]
ExecStart=/home/pi/formide-client/bin/formide-client
Restart=always
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=formide-client
User=pi
Group= pi
Environment=NODE_ENV=production
WorkingDirectory=/home/pi/formide-client

[Install]
WantedBy=multi-user.target
```

Then run `systemctl enable formide-client` to have the service run on boot.

Let me know if this works for you!

shaiss, 2017-02-23:
tried that no luck.  I checked with netstat for open ports and grep'd PS -aux for formide with no avail.

ChrisTerBeke, 2017-02-23:
Maybe there's a conflict now since `/etc/init.d/formideboot.sh` is also still activated? `Systemd` services shouldn't need any external references or trigger from scripts. The `formide-client.service` file should be all that's needed to get it to work. 

`/home/pi/formide-client/bin/formide-client: No such file or directory` is another issue though, which I'll take a look at tomorrow morning to get the correct file path instead.

ChrisTerBeke, 2017-02-24:
I've created an updated `systemd` file and tested it on my own Pi. 

```
[Unit]
Description=Formide Client

[Service]
ExecStart=/home/pi/.nvm/versions/node/v4.7.0/bin/node /home/pi/.nvm/versions/node/v4.7.0/lib/node_modules/formide-client/app.js
Restart=always
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=formide-client
User=pi
Group= pi
Environment=NODE_ENV=production

[Install]
WantedBy=multi-user.target
```

I actually want to improve the global script that `npm` links to when installing `formide-client` globally to be able to make the `systemd` file better, as the `ExecStart` command is a bit messy now because of using `nvm`.

shaiss, 2017-02-24:
WOOT, that worked!!!  Nice work!  now I can run longer prints, once the annoying first layer fan thing is fixed :P

ChrisTerBeke, 2017-02-24:
Great! I'll add this script to the repo for the next release of `formide-client` and have it copy automatically at the end of the install script, as I image most people would wanna run the client on boot anyways. The fan thing is already being worked on :)

ChrisTerBeke, 2017-03-24:
Auto boot and webcam setup and scripts are now fully documented in this project's wiki, closing this issue.

closed 2017-03-24 by ChrisTerBeke