PRINTR3D/formide-client #287: Create containerized version using Docker

fonix232, 2017-04-06:
It would make things a lot easier for many people if a containerized version was made available (also would make installation easier on a RasPi!).

I, for one, would like to use my existing Synology NAS to act as a client, with the printer connected to it. I've already tried making a docker image using the official Node base images, however installation fails on the formide-drivers package during node-pre-gyp (or if node-gyp is installed manually, it fails on epoll).

ChrisTerBeke, 2017-04-06:
Hey @fonix232,

Thanks for the request! Dockerizing the Formide client would indeed be handy for running it in other environments than The Element or Raspberry Pi. There are currently 2 things that need to happen before this setup will work:

1) At this moment our drivers are only compatible with ARM architecture processors, so in order to make this work on machines like your NAS we'll need to compile an x86 version of the drivers as well when we make releases. I'll discuss with our engineers tomorrow how much time it would take to set up automated builds for x86 as well.

2) The `epoll` package is currently part of the core client dependencies to get USB host functionality, but in client v2 (https://github.com/PRINTR3D/formide-client-2) this will be a plugin that's only needed in some of our manufacturer integrations. V2 is currently in beta and will be rolled out to Elements and other devices soon.

Hope we can help you to have it up and running soon!

fonix232, 2017-04-06:
In the meantime, I've created a Dockerfile that actually works. It relies, however, on a Jessie-based NodeJS image, which means it's over 750MB.

If it was possible to compile epoll on, say, an alpine-based image, it would work wonders.

And about drivers - how about using the native code compiling features of NodeJS, instead of shipping ready-to-use binaries? Or are the driver sources proprietary?

Also, sidenote. With the current system requiring the whitelisting of MAC addresses that can connect online, Docker only helps to solve one side of the problem. I'd kindly suggest using an API key system, or even auto-authorizing devices when an active user account logs in on it. What's the point of pre-authorizing the devices any way?

fonix232, 2017-04-06:
[Dockerfile](https://github.com/PRINTR3D/formide-client/files/903737/Dockerfile.txt)

Here is the Dockerfile I made. Nothing much, but took quite a bit of testing to get it working. Working, as in, it builds, boots, and makes the web UI accessible.

ChrisTerBeke, 2017-04-06:
The driver source is currently closed, that's why we only have pre-built binaries using node-pre-gyp available.

With regards to the whitelisting, this is in place because previously only The Element and integrated printers were able to access the platform. Now that Raspberry Pi (and other devices) come into play, we're working on opening up this system. When client v2 is ready for public usage, that will also the point where connecting to Formide will not require pre-whitelisting anymore. If you wanna get online with your current NAS setup, please send me an email on chris@printr.com to get the MAC address whitelisted.

Also thanks for the Dockerfile, we'll definitely be testing it out! When epoll is removed from the base client, and the drivers are working on x86 as well, making an Alpine based image should be very doable!

fonix232, 2017-04-06:
Making an alpine base image will literally only require the following Dockerfile:

    FROM node:4-alpine
    RUN npm install -g node-pre-gyp
    RUN npm install -g formide-client
    EXPOSE 8080
    CMD formide-client

This, if you want to install from NPM. If you want to bundle the current source code, then just dump the following Dockerfile next to the package.json:

    FROM node:4-alpine
    EXPOSE 8080

Building this Dockerfile with the project around it will include it in the image itself and run "npm start".

ChrisTerBeke, 2017-04-06:
Hey. I've started putting together a Dockerfile based on Alpine that installs all the needed dependencies on the client V2 repo. Apart from needing a linux x86/x64 version of the drivers and proper device implementation configuration it works. It's currently in a separate branch over at https://github.com/PRINTR3D/formide-client-2/blob/feature-docker/Dockerfile.

ChrisTerBeke, 2017-04-20:
Hey @fonix232,

We have just finished our automated setup for x86 driver builds. It is released with client v2 (2.0.7). Please check out PRINTR3D/formide-client-2#master to get the correct version that is Docker compatible!

Please let us know if you have any more issues with running in Docker by opening an issue on the client v2 repository, thanks in advance!

closed 2017-04-20 by ChrisTerBeke

ChrisTerBeke, 2017-04-20:
We've published formide-client to Docker's repository store as well, making it available for anyone using the Docker CLI. Just run `docker run printr/formide-client` to pull and run it! More info at https://store.docker.com/community/images/printr/formide-client.